clade/PyDAQmx #24: nicaiu.dll fails to import with NI-DAQmx 15

jlien-sla, 2015-09-10:
Recently tried to upgrade to NI-DAQmx 15.0.1 and my application stopped working.  When I try to 'import pyDAQmx' it fails on loading the nicaiu.dll, saying the DLL failed to initialize (WinError 1114).

Previous version of NI-DAQmx was 14.5.1 which works fine.  OS is Win 7 Pro 64 bit.

clade, 2015-09-11:
My guess is that the error comes from the system and not from PyDAQmx.

* Have you uninstaled the previous version ?

* Is you new installation of NIDAQmx working (for example are you able
to test your instrument with NI MAX) ?

* You need to use the same architecture for NIDAQmx and Python
(probably 64bits).

* Do you get the same error if you try to compile and use an ANSI C
example for National Instruments ? If it works then there is
a problem in PyDAQmx. If it doesn't then you should report the problem 
to NI.

Hope this will help you to solve your problem.

jlien-sla, 2015-09-11:
You are correct it was a problem with the MAX installation.  Sorry for the false alarm, and thanks for your help and a great library.

closed 2015-09-11 by jlien-sla