clade/PyDAQmx #25: Array type issue at import when using PyPy

Apteryks, 2015-09-24:
Here is the traceback:
[PyPy 2.6.1 with MSC v.1500 32 bit] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>>
>>>> import PyDAQmx
C:\Users\maco\virtualenvs\pypy\site-packages\numpy\linalg\lapack_lite.py:78: UserWarning: tuned lapack (openblas, atlas ...) not found, using lapack_lite
  warnings.warn('tuned lapack (openblas, atlas ...) not found, using lapack_lite')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\maco\virtualenvs\pypy\site-packages\PyDAQmx\__init__.py", line 7, in <module>
    from DAQmxFunctions import *
  File "C:\Users\maco\virtualenvs\pypy\site-packages\PyDAQmx\DAQmxFunctions.py", line 146, in <module>
    array_type(_type),2) for _type in type_list_array]
  File "C:\Users\maco\virtualenvs\pypy\site-packages\PyDAQmx\DAQmxFunctions.py", line 111, in array_type
    flags=('C_CONTIGUOUS','WRITEABLE'))
  File "C:\Users\maco\virtualenvs\pypy\site-packages\numpy\ctypeslib.py", line 253, in ndpointer
    elif isinstance(flags, flagsobj):
TypeError: isinstance() arg 2 must be a class, type, or tuple of classes and types

I'm using the Windows binary release of Pypy and installed numpy using: 'pip install git+https://bitbucket.org/pypy/numpy.git'. PyDAQmx was installed using 'pip install PyDAQmx' (v 1.3.1).

clade, 2015-09-24:
>                 File "", line 1, in

Your error comes from the following command, which works well on C Python.

numpy.ctypeslib.ndpointer(numpy.int32, flags=('C_CONTIGUOUS', 'WRITEABLE'))

It seams to me that it is an error from the PyPy fork of numpy and not 
from PyDAmx. You should try to ask them.

rlamy, 2015-09-24:
This particular issue should be solved in numpypy now, cf. https://bitbucket.org/pypy/numpy/commits/6b5083ac1859da3fab0e6229f8367ad544239896. However, I have no idea whether it's enough to get PyDAQmx working.

Apteryks, 2015-09-24:
@clade, thank you for your answer. You are right, this is on PyPy numpy side rather than in PyDAQmx. Feel free to close this issue!

closed 2015-09-25 by clade