clade/PyDAQmx #30: Remove DAQmx_ prefix from module

eric-wieser, 2016-07-19:
That way instead of doing this and causing arbitrary namespace pollution:

```python
from PyDAQmx import *

DAQmx_Some_Function(DAQmx_SomeConstant)
```

We can use the safer

```
impot PyDAQmx

PyDAQmx.Some_Function(PyDAQmx.SomeConstant)
```

petebachant, 2016-07-20:
Great idea. Better yet, make the names comply with PEP8 so they're not so hard to type!

clade, 2016-07-20:
Hi ! 

In the dev branch, I modify the __init__ to add the name without the prefix. This modification is backward compatible (the from PyDAQmx import * will import the name with the prefix). Can you test it !

I hesitate to rename the function using the PEP8 for two reasons : the documentation is not available for Python but in C. The second reason is that functions are closer to C functions than to python functions. I have in mind to write real python function (for example to remove the byref and returns the value). In this case I will rename the function using the PEP8

Pierre

eric-wieser, 2016-07-20:
> In the dev branch, I modify the init to add the name without the prefix. 

Looks good to me!