clade/PyDAQmx #32: pep8-ify function names

eric-wieser, 2016-07-20:
Following on from [this comment](https://github.com/clade/PyDAQmx/issues/30#issuecomment-233812704).

So instead of

```python
task.CfgSampClkTiming(
	source=None,
	rate=1,
	activeEdge=daq.Val_Rising,
	sampleMode=daq.Val_FiniteSamps,
	sampsPerChan=1
)
```

it would become

```python
task.cfg_sample_clk_timing(
	source=None,
	rate=1,
	active_edge=PyDAQmx.VAL_RISING,
	sample_mode=PyDAQmx.VAL_FINITE_SAMPS,
	samps_per_chan=1
)
```

This potentially raises the issue that the module name should be PEP8'd as well, becoming the much easier to type `pydaqmx`

eric-wieser, 2016-07-20:
@clade's remark from that thread:

> I hesitate to rename the function using the PEP8 for two reasons : the documentation is not available for Python but in C. The second reason is that functions are closer to C functions than to python functions. I have in mind to write real python function (for example to remove the byref and returns the value). In this case I will rename the function using the PEP8

petebachant, 2016-07-20:
Maybe for the next major release, the package name should also be changed to `pydaqmx` for similar reasons.

eric-wieser, 2016-07-20:
That needn't wait - the switch could be made right now, and a reasonably simple `PyDAQmx.py` containing `from pydaqmx import *` added

clade, 2016-07-21:
The dev2 branch is a tentative to pep8-ify the package. 

The setup.py should install two package : the main one (pydaqmx) and a legacy one (PyDAQmx) that depends on the first one. 

Changing the name of the all the functions, constant and arguments was not easy so I have decided to completely change the structure of the package and to use object to create the functions. 

You can see the functions and constants with the commands : 

    import pydaqmx
    print pydaqmx.functions.__all__
    print pydaqmx.constants.__all__
    print pydaqmx.cfg_samp_clk_timing.__doc__

clade, 2016-07-22:
The dev2 branch is now tested.

For example : 

```
with pydaqmx.Task() as t:
    t.create_ai_voltage_chan("dev1/ai0","",pydaqmx.VAL_CFG_DEFAULT,-10.0,10.0,pydaqmx.VAL_VOLTS,None)
```