clade/PyDAQmx #35: Convert all the DAQmxError.* constants into DAQmxError subclasses

eric-wieser, 2016-08-03:
Something like:

```python
for c in constant_list:
    if not c.startswith('DAQmxError'):
        continue
    code = getattr(daq_constants, c)
    errname = c[10:]
    if code < 0:
        globals()[errname + 'Error'] = type(errname + 'Error', (DAQmxError,), dict(code=code})
    else:
        globals()[errname + 'Warning'] = type(errname + 'Warning', (DAQmxWarning), dict(code=code})
```

clade, 2016-09-06:
Commit https://github.com/clade/PyDAQmx/commit/b6679c9 solve this issue. 

Errors are defined in the `DAQmxFunctions` module and can be imported from the PyDAQmx package (for example `PyDAQmx.InvalidDeviceIDError`).

eric-wieser, 2016-09-06:
Nice!

closed 2016-09-06 by clade