clade/PyDAQmx #46: Problems with Windows 10 64-bit

jhyearsley, 2017-06-28:
## First issue:
``` python
NotImplementedError: Location of niDAQmx library and include file unknown on win32 - if you find out, please let the PyDAQmx project know
```
I have found the locations.
The niDAQmx library and include files are located in the following paths:
``` python
lib_name = 'C:/Windows/System32/nicaiu.dll'
dot_h_file = 'C:/Program Files (x86)/National Instruments/Shared/ExternalCompilerSupport/C/include/nisyscfg.h'
```
## Second issue:
After first issue has been solved (i.e. I have manually inserted the correct paths) there is an error within the `Task.py` file.
``` python

  File "C:\Users\johny\AppData\Local\Continuum\Anaconda2AGAIN\lib\site-packages\PyDAQmx\Task.py", line 142, in Task
    del function_name, name, func, arg_names, doc

NameError: name 'function_name' is not defined
```
It appears that `function_dict` and `function_list` are not being populated. 

## Takeaway
The package does not work with Windows 10 as of now and I am not familiar enough to debug any further. Also as a side note I think the codebase would be much better off if some of the importing conventions were changed. For instance imports like this make everything pretty confusing
``` python
import DAQmxFunctions
from DAQmxFunctions import *
```

clade, 2017-06-28:
Are you sure that NI-DAQmx is install on your computer. There should be a file called NIDAQmx.h somewhere. I don't think its name has changed.

jhyearsley, 2017-06-29:
You were correct, I had to install an earlier version of the NI-DAQmx driver to find the NIDAQmx.h file. Thanks!

closed 2017-06-29 by jhyearsley