clade/PyDAQmx #48: [information only] NI now has an official python repo

iamed18, 2017-09-04:
It should probably be mentioned in README.md that NI now has an official Python API on their Github. It can be found here:

https://github.com/ni/nidaqmx-python/

AlexJaeger, 2018-02-16:
Sounds like a good idea. It would be good to also clarify that according to NI, nidaqmx only supports the full NIDAQmx on Windows (i.e. no support for NIDAQmxBase on Linux/Mac)

For crossplatform support, PyDAQmx is still the way to go.

albertogomcas, 2022-06-13:
I decided some years go to not migrate towards the NI wrapper because they spent years (from 2017) without a release, whereas PyDAQmx remained active. 

But after 5 years of looking like a dead repo it seems the official NI library picked up speed again and there have been 5 new releases during 2022. In particular Linux is officially supported from 0.6.1.