clade/PyDAQmx #49: Issues using PyDAQmx 1.3.2 with 64bit Python 3.6.3 on Centos 7

deveshks, 2017-11-09:
Hi,

I tried installing PyDAQmx 1.3.2 on my Centos  7.4.1708 running on an 64bit Intel mini-PC. I am using NI-DAQmx Base 15.0 for Linux, and I am running a 64-bit Python 3.6.3. I am able to successfully install PyDAQmx using pip, but I get the following error when I import it.

Python 3.6.3 (default, Oct 11 2017, 18:17:37) 
[GCC 4.8.5 20150623 (Red Hat 4.8.5-16)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyDAQmx
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tesla/.local/lib/python3.6/site-packages/PyDAQmx/__init__.py", line 7, in <module>
    from .DAQmxFunctions import *
  File "/home/tesla/.local/lib/python3.6/site-packages/PyDAQmx/DAQmxFunctions.py", line 86, in <module>
    DAQlib, DAQlib_variadic = DAQmxConfig.get_lib()
  File "/home/tesla/.local/lib/python3.6/site-packages/PyDAQmx/DAQmxConfig.py", line 62, in get_lib
    ctypes.CDLL('/usr/local/lib/liblvrtdark.so', mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib64/python3.6/ctypes/__init__.py", line 343, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/local/lib/liblvrtdark.so: wrong ELF class: ELFCLASS32
>>> 

Has this problem been seen before, and if yes, is there a way to resolve this.

Best,
Devesh

deveshks, 2017-11-11:
I was able to rectify this issue by replacing line #62 in https://github.com/clade/PyDAQmx/blob/master/PyDAQmx/DAQmxConfig.py

from  ctypes.CDLL('/usr/local/lib/liblvrtdark.so', mode=ctypes.RTLD_GLOBAL)
to ctypes.CDLL(find_library('lvrtdark'), mode=ctypes.RTLD_GLOBAL)

I also verified this by running the last example on https://pythonhosted.org/PyDAQmx/usage.html using a USB-6210 NI DAQ. Let me know if you want me to make those changes in the library.

Best,
Devesh

clade, 2017-11-14:
Thanks for your fix : commit 6f8d3a0

closed 2017-11-14 by clade