clade/PyDAQmx #52: `__version__` string did not change from v1.4 to v1.4.1

chrisjbillington, 2018-11-13:
I have code that relies on functionality in PyDAQmx v1.4.1, and my code checks that the PyDAQmx version is new enough so that it can raise an exception telling the user to update PyDAQmx.

However, `PyDAQmx.__version__` is `'1.4'`, on version 1.4.1.

The version string should be updated to 1.4.1. Or, I suppose, 1.4.2 since you would need to release a new version to include the fix!

clade, 2018-11-13:
Thanks for you message. I release the version 1.4.2 according to your suggestion. The version string is in two places (the setup.py, which was correct, and the __init__.py). 
Usually, I import the package in the setup.py to get the version, but it is not possible for PyDAQmx as the package cannot be imported if NIDAQmx is not installed. 
I have to figure out a solution.

chrisjbillington, 2018-11-13:
Thanks very much :)

For my projects, I include the version only in setup.py, and when setup.py runs, I have it generate a `__version__.py` within the package directory. Then I have `__init__.py` import the version from there, or set `__version__` to None if it cannot. [Here is an example](https://github.com/chrisjbillington/inotify_simple/blob/master/setup.py). I'm not sure if this is optimal, but it has worked for me so far.

closed 2019-02-24 by chrisjbillington