clade/PyDAQmx #54: PyDAQmx 1.4.2 doesn't install correctly on Python 2.7

philipstarkey, 2019-04-11:
I've just tried installing PyDAQmx on Python 2.7.16 (a new conda 2.7 environment) using pip and it is not installing correctly (on Windows 10 anyway).

It seems to be installing into `site-packages/pydaqmx` rather than `site-package/PyDAQmx`. This means that code using PyDAQmx obviously breaks as you can't import PyDAQmx. I tried `import pydaqmx` but this throws an Exception because it can't import `Task`.

PyDAQmx==1.4.1 works correctly, so hopefully it is easy to work out what caused this change!

clade, 2019-04-11:
How did you installed PyDAQmx ?

The pydaqmx is the PEP8 compliant version PyDAQmx which is in the dev2 branch (and supposed to be released as version 2.xx).

philipstarkey, 2019-04-11:
Installed using pip. Possibly the .whl on PyPi for Python 2.7 was built using the wrong branch?

philipstarkey, 2019-04-12:
I looked into this a bit further and it seems v1.4.2 is the first release that I can see which includes a .whl on PyPi for Python v2 (see https://pypi.org/project/PyDAQmx/1.4.2/#files )

If you download the .whl and open it, it appears to contain a weird mixture of files/folders from **both** the master and dev2 branch. No idea how that happened. 

Given that this is the first time a .whl has been uploaded to PyPi, I presume if you just delete it then it will build from the .tar.gz source (by running setup.py) on Python 2.7 like before, and everything will be fine.