clade/PyDAQmx #57: NIDAQmx now supported on Linux?

bburan, 2019-10-08:
It seems that NI may now be providing support for the full version of NIDAQmx on Linux based on these instructions (http://www.ni.com/product-documentation/54754/en/). After following these instructions on CentOS 8, I installed PyDAQmx and got the error about being unable to find the NI-DAQmx headers. However, when I updated the `dot_h_file` to point to `'/usr/include/NIDAQmx.h'`, it loaded properly.

bburan, 2019-10-10:
I can confirm that after making the change to the `dot_h_file` path described in the first comment, my application (https://github.com/bburan/psiexperiment) runs fine on Linux. To give you an idea of how heavily I rely on the NIDAQmx library, take a look at this file (https://github.com/bburan/psiexperiment/blob/master/psi/controller/engines/nidaq.py). I didn't have to do anything else to get my code working on Linux.

clade, 2019-10-10:
Thank you for your comment.
Can you test the dev branch on your system https://github.com/clade/PyDAQmx/blob/dev/PyDAQmx/DAQmxConfig.py
lin 46 to 50

I check for both path (the old and the new one).

bburan, 2019-10-16:
I can confirm it works.

bburan, 2019-11-26:
Are there any plans to release a new version with support for Linux?

clade, 2020-01-12:
Version [1.4.3](https://github.com/clade/PyDAQmx/tree/v1.4.3) available on [github](https://github.com/clade/PyDAQmx/tree/v1.4.3) and [Pypi](https://pypi.org/project/PyDAQmx/).

closed 2020-01-12 by clade