clade/PyDAQmx #58: No attribute WriteAnalogScalarF64

marcodeltutto, 2020-02-26:
Hello, I am on macOS with python3, and if I try to run example2.py I get the following error:
```
Traceback (most recent call last):
  File "PyDAQmx/test.py", line 20, in <module>
    task_out.WriteAnalogScalarF64(1,10.0,value,None)
AttributeError: 'Task' object has no attribute 'WriteAnalogScalarF64'
```

Any suggestions on how to fix it? Thank you!

clade, 2020-02-27:
On macOS, you are using the nidaqmxbase driver and not the nidaqmx one. Only a subset of functions are available, and WriteAnalogScalarF64 is not. In your case you should probably put your value in an array and use task.WriteAnalog64 

```
import numpy as np
data = np.array([value])
n=1
sampsPerChanWritten=int32()

task.WriteAnalogF64(n, 0, 10.0, DAQmx_Val_GroupByScanNumber, data, 
     byref(sampsPerChanWritten), None)

```

marcodeltutto, 2020-02-27:
Thank you, it works!